pixhawk/Pixhawk-Standards #8: DS-009 Revision: 0.1.0 Page 9 has ambiguities in regards Analog Power connector.

ApiumJacob, 2020-08-19:
"The VCC lines have to offer at least 2.5A continuous and should default to 5.3V."  

1.  There are two lines.  Should each line offer at least 2.5A or show the all lines combined offer 2.5A?  I suspect the latter as the GH connectors are only rated for 1.0A per contact.

2. It is not clear who is offering the 2.5A.  Is this coming out of this connector or going into it or maybe a little of both?

mrpollo, 2020-09-29:
Thanks @ApiumJacob we will revise the doc and add a note to clear out the page, the answer is 2.5A total

mrpollo, 2020-09-29:
Ping @davids5

davids5, 2020-10-03:
The Spec need the connector labeled as POWER input and directions noted.

@LorenzMeier - GH spec is here https://www.jst-mfg.com/product/pdf/eng/eGH.pdf #26 GA is max 1 A. per contact.