pixhawk/Pixhawk-Standards #9: DS-009 Revision: 0.1.0 General comment on signal direction and pin intentions.

ApiumJacob, 2020-08-19:
Page 5 Telemetry Port

o. Is VCC intended to provide power to the attached device or to provide power to the pixhawk device?
o. If intended to provide power to attached device is the maximum allowable current draw?
o. If intended to provide power to power pixhawk device what is the expected current to be sourced?
o. Is there a specific bit rate, number of bits, parity or number of stop bits for the telemetry or should it be configurable?
o. Is XON/XOFF supported?

Page 6 Full GPS plus Safety Switch Port

o. Same issue with VCC as on page 5.
o. I2C pins labeled as (IN/OUT).
o. Do I2C need pullup's, implied that they do since the I2C port on page 8 explicitly says where the pullups sit.
o. SAFETY SWITCH, place IN in parenthesis for consistency: (IN).
o. SAFETY LED, assuming this is an (OUT).
o. VDD_3V3, if this powers the LED, why is "SAFETY LED" +3.3V and not GND?
o. What is the expected maximum power output from this pin can drive, assuming it is an output.

Page 7 Basic GPS port

o. Same issue with VCC as on page 5.
o. I2C pins labeled as (IN/OUT).
o. Do I2C need pullup's, implied that they do since the I2C port on page 8 explicitly says where the pullups sit.

Page 7 CAN

o. Same issue with VCC as on page 5.
o. CAN pins labeled as (IN/OUT).
o. Are can pins expected to go to an external CAN bus driver or are these actual CAN bus signals coming from an internal driver?  The note on using a recommended CAN transceiver implies that you need to add them, but the Volt field of CAN high and CAN low implies that they are CAN bus levels.

Page 8 I2C Port

o. Same issue with VCC as on page 5.
o. I2C pins labeled as (IN/OUT).

Page 8 SPI Port

o. Same issue with VCC as on page 5.
o. Assuming pixhawk is master (explicitly say this)
o. If above assumption is correct the SS lines are (OUT)

Page 8/9 Analog Power

With regard to the CURRENT and VOLTAGE signals:

o. Are these inputs or outputs?  Assuming (IN).
o. What is the absolute minimum and maximum for these signals?

mrpollo, 2020-09-29:
Hey @davids5 this might be something you can help us with.

davids5, 2020-10-03:
> Page 5 Telemetry Port
> 
> o. Is VCC intended to provide power to the attached device or to provide power to the pixhawk device?
It is a supply to the device.

> o. If intended to provide power to attached device is the maximum allowable current draw?

Telem 1 is 1.5A max by its self , all others connectors  combined are 1.5 A 

> o. If intended to provide power to power pixhawk device what is the expected current to be sourced?

No. 

> o. Is there a specific bit rate, number of bits, parity or number of stop bits for the telemetry or should it be configurable?

Configurable, See QGS http://qgroundcontrol.com/

> o. Is XON/XOFF supported?

@dagar?
 
> Page 6 Full GPS plus Safety Switch Port
> 
> o. Same issue with VCC as on page 5.

This is Power to device.

> o. I2C pins labeled as (IN/OUT).

The FMU is the master. The slaves on connected here. Clock (SCL) is an out from FMU SDA is bi-directional.

> o. Do I2C need pullup's, implied that they do since the I2C port on page 8 explicitly says where the pullups sit.

Pull ups are on master (FMU)

> o. SAFETY SWITCH, place IN in parenthesis for consistency: (IN).

It is  actually `SAFETY SWITCH IN` There is a 1.5 K to gnd on the FMU

The other end of the SW is on V3.3 MAX). SW closed => 3.3V on `SAFETY SWITCH IN`

> o. SAFETY LED, assuming this is an (OUT).

Yes 

> o. VDD_3V3, if this powers the LED, why is "SAFETY LED" +3.3V and not GND?

provides ground to a 68 Ohm resistor connected to cathode of LED anode is to VDD 3v3 

> o. What is the expected maximum power output from this pin can drive, assuming it is an output.

~ 20 mA

> 
> Page 7 Basic GPS port
> 
> o. Same issue with VCC as on page 5.

it is an out it is a member of the 1.A peripheral power node. 

> o. I2C pins labeled as (IN/OUT).

Same as above.

> o. Do I2C need pullup's, implied that they do since the I2C port on page 8 explicitly says where the pullups sit.

Same as above.

> 
> Page 7 CAN
> 
> o. Same issue with VCC as on page 5.

it is an out it is a member of the 1.A peripheral power node. 

> o. CAN pins labeled as (IN/OUT).

It is a bus.  FMU is providing 120 ohm temination. 

> o. Are can pins expected to go to an external CAN bus driver or are these actual CAN bus signals coming from an internal driver? The note on using a recommended CAN transceiver implies that you need to add them, but the Volt field of CAN high and CAN low implies that they are CAN bus levels.

CAN TX/RX are the TTL LS side of a CAN transceiver
CAN_H, CAN_L are the CAN bus side of a CAN transceiver 

So this is the CAN bus side of a CAN transceiver 

> 
> Page 8 I2C Port
> 
> o. Same issue with VCC as on page 5.

Same Answer.

> o. I2C pins labeled as (IN/OUT).

Same Answer.

> Page 8 SPI Port
> 
> o. Same issue with VCC as on page 5.

Same Answer.

> o. Assuming pixhawk is master (explicitly say this)

Yes.

> o. If above assumption is correct the SS lines are (OUT)

Yes.

> Page 8/9 Analog Power
> 
> With regard to the CURRENT and VOLTAGE signals:

Theses are ADC inputs from a Pixhawk Power ANALOG module.  

google (pixhawk power module schematic)

[image]

Scaled to 3.3V

> o. Are these inputs or outputs? Assuming (IN).

Ins.

> o. What is the absolute minimum and maximum for these signals?

See https://github.com/pixhawk/Hardware 

https://github.com/pixhawk/Hardware/blob/master/FMUv2/PX4FMUv2.4.5.pdf

for the signal conditioning 

The F7 uses 1M, 1nF  to gnd and 10K series to the ADC channels.