pixhawk/Pixhawk-Standards #31: Standardize Digital Power Port for Pixhawk Connector Standard

vincentpoont2, 2022-12-13:
[The JST GH](https://web.archive.org/web/20191003020420/http://www.jst-mfg.com/product/detail_e.php?series=105) in the connector standard is only rated for 1A per pin. Most flight controller can have power input up to 3A. This is the reason Holybro changed the power ports connectors on the Pixhawk 5X & 6X to [Molex CLIK-Mate](https://www.molex.com/molex/products/part-detail/pcb_receptacles/5024430760). ARK and CubePilot also uses this connector for power.  

I suggest that we standardize this port for Digital Power Ports.

vincentpoont2, 2022-12-13:
@mrpollo What do you think?

mrpollo, 2022-12-13:
Hey @davids5 what do you think?

davids5, 2022-12-13:
@mrpollo - Technically I think it is a good change. But I may not have a complete picture. Please send a poll to the complete membership and see if there are any objections.