pixhawk/Pixhawk-Standards #35: DS-010 PAB X2 Spare pin assignment

PetervdPerk-NXP, 2023-02-23:
As discussed in the [Hardware dev call](https://discuss.px4.io/t/hardware-pixhawk-dev-call-feb-23-2023/30814)

We propose an update of DS-010 to assign some applications to the spare pins on the X2 PAB connectors.

We want to introduce 3 extra applications

- Ethernet RX_ER and PHY_INT pins
- CAN3 
- 4 Extra channels from the FMU

Which would result in the following pinout of PAB X2 changes are in **bold**
Pin   Name | Label | Voltage | Direction
-- | -- | -- | --
Pin   2 | ETH_MDIO | 3V3 | Input/Output
Pin   4 | ETH_MDC | 3V3 | Output
Pin   6 | ETH_POWER_EN| 3V3 | Output
Pin   8 | GND | 0V |  
**Pin   10** | ETH_RX_ER | 3V3 | Output
**Pin   12** | ETH_PHY_nINT | 3V3 | Input
**Pin   14** | GND | 0V |  
**Pin   16** | FMU_CH9 | 3V3 | Output
**Pin   18** | FMU_CH10 | 3V3 | Output
**Pin   20** | FMU_CH11 | 3V3 | Output
**Pin   22** | FMU_CH12 | 3V3 | Output
**Pin   24** | GND | 0V |  
Pin   26 | SPARE09 | NC  |  
Pin   28 | SPARE10 | NC|   |  
Pin   30 | SPARE11 | NC|   |  
Pin   32 | SPARE12 | NC|   |  
Pin   34 | SPARE13 | NC|   |  
Pin   36 | SPARE14 | NC|   |  
**Pin   38** | GND | 0V |  
**Pin   40** | CAN3_TX | 3V3 | Output
**Pin   42** | CAN3_RX | 3V3 | Input
Pin   44 | PG6 | 3V3 | Input/Output
Pin   46 | GND | 0V  |  
Pin   48 | NFC_GPIO | Input/Output  |  
Pin   50 | GND | 0V  |

davids5, 2023-02-23:
@PetervdPerk-NXP Please add columns for `was` and `is` to the table so all the information is in one place

From the  FMU side for ETH_RX_ER and  ETH_PHY_nINT are inputs

doblerc, 2023-03-09:
I have no issue with assigning these pins to proposed usage. These pins just need to be clearly marked as optional. Otherwise all STM32 based FMUs loose their compliance with the PAB standard.

davids5, 2023-03-09:
✅ Nscdg approves this change

igalloway, 2023-03-10:
During the meeting it was asked why do we need the ETH_RX_ER and ETH_PHY_nINT if they are not being used in the code yet, and there is possibility to use in band signaling for the ER (error)

I will note here: we do intend to use this hardware connection between NXP processors and the NXP PHYs in the future. Hardware based irq and error feedback can become quite important for enhanced reliability i.e. ROS2 interfacing

vincentpoont2, 2023-03-14:
✅ Holybro approves this change

> I have no issue with assigning these pins to the proposed usage. These pins just need to be clearly marked as optional. Otherwise all STM32 based FMUs loose their compliance with the PAB standard.

I agree it should be mentioned somewhere in the doc, preventing all the current FMU from "losing compliance".

igalloway, 2023-05-17:
@mrpollo  - What is needed to get this PR approved?

mrpollo, 2023-05-17:
✅ Dronecode approves this change

mrpollo, 2023-05-17:
Hey @igalloway @dk7xe thanks for the ping, we are moving forward with the proposal.

@davids5, can you please update the main sheet with the proposal from NXP? I will update the DS-010 doc and push to github as PDF

davids5, 2023-05-17:
@mrpollo  Yes. I will get an updated schematic to you.

davids5, 2023-05-18:
@mrpollo - The Table and the Document are updated with the RC03 Pinning

davids5, 2023-06-01:
Reference Documents are updated and upload 

[image]

@mrpollo Please close this. 

Thank you @PetervdPerk-NXP

closed 2023-06-01 by mrpollo