pixhawk/Pixhawk-Standards #39: SPI Pinout on Pixhawk Connector Standard

vincentpoont2, 2023-05-19:
Does the SPI port need to be updated?

Currently, the [DS-009 Pixhawk Connector Standard.pdf ](https://github.com/pixhawk/Pixhawk-Standards/blob/master/DS-009%20Pixhawk%20Connector%20Standard.pdf) shows 7 pin on the SPI port as follow

Pin | Signal | Volt
-- | -- | --
1(red) | VCC | +5V
2 (black) | SPI_SCK | +3.3V
3(black) | SPI_MISO | +3.3V
4(black) | SPI_MOSI | +3.3V
5(black) | !SPI_SS1 | +3.3V
6(black) | !SPI_SS2 | +3.3V
7(black) | GND | GND

However, all the FMUv6X hardware has SPI port that uses 11pin as follow:

Pin | Signal | Volt
-- | -- | --
1(red) | VCC | +5V
2 (black) | SPI6_SCK | +3.3V
3(black) | SPI6_MISO | +3.3V
4(black) | SPI6_MOSI | +3.3V
5(black) | SPI6_CS1 | +3.3V
6(black) | SPI6_CS2 | +3.3V
7(black) | SPIX_SYNC | +3.3V
8(black) | SPI6_DRDY1 | +3.3V
9(black) | SPI6_DRDY2 | +3.3V
10(black) | SPI6_nRESET | +3.3V
11(black) | GND | GND

vincentpoont2, 2023-05-19:
@mrpollo @davids5 
Should we update this?

davids5, 2023-05-19:
I think is should be added.

davids5, 2023-05-19:
@mrpollo Shall we add this to the Spec?

mrpollo, 2025-01-13:
This was resolved

closed 2025-01-13 by mrpollo