pixhawk/Pixhawk-Standards #42: Hardware Rev and Ver for new FMUV6X Board

XXGurjot07, 2023-09-13:
I'm designing a new FMUV6X Based FMU+IO Unit integrated with a Linux machine and LTE communication stack.
How should I configure FMU_REV and FMU_VER on this board?

I can see some of these are free, how should I get it allocated to my design if possible?
is VER 6 alright?
is REV 5 alright?
And how do these play out in software when building and running PX4?

davids5, 2023-09-14:
Will you be running your own board configuration or expecting to use the `px4_fmu-v6x` build?

If it is the former then you have complete freedom to use any code points you wish or not even support the ver and rev.

If it is the latter, you must do a pull request against https://github.com/pixhawk/Pixhawk-Standards/blob/master/DS-019%20Pixhawk%20versions%20and%20revisions.md to reserve the values. Followed by a PR to PX4

XXGurjot07, 2023-09-17:
Thanks, typically I would like this to use `px4_fmu-v6x`.
However I'm combining FMU+BASEBOARD+LINUX/LTE on one board and vibrationally insulated IMU unit on another board.
So should I do a pull request?