pixhawk/Pixhawk-Standards #55: DS012- V6X standard, which are 3 pins  (GPIO) for bootloader?.

Daywithck, 2024-09-17:
Hi,
     I can see the in block diagram(DS-012 V6X standard) ADIO(8pin JST GH)  3 bits for bootloader.  I couldn't figure out corresponding 3pins used for boot loader in pinout sheet.  but i can guess it like, VBUS_Sense, may be nArmed, ?.   if any one knows, pls share the pin names.  Appreciate your inputs.

davids5, 2024-09-17:
[image]

There is some logic on the base. 
FMU_nRST (pin 87 on X1) = !nArmed (Pin 85 on X1)&  FMU_RST_REQ_AUX 
VBUS_SENSE (pin 80 on X1)  = FMU_BOOTLOADER_AUX || VBUS

Daywithck, 2024-09-18:
make sense. thanks for the update.

closed 2024-12-09 by mrpollo