pixhawk/Pixhawk-Standards #58: Clarification on resistor values for encoding the ver/rev codes

KarthiAru, 2024-12-09:
Hi Team,

I need clarification on the resistor values to use for encoding the ver/rev codes to the value 0x007.

[image]

Based on the voltage divider rule and the table referenced in the image above, I calculated:

- R1 = 38.3kΩ
- R2 = 115kΩ

These values correspond to a code of 0x007 for the specified version/revision.

However, in a [recent PR](https://github.com/pixhawk/Pixhawk-Standards/pull/53), it was suggested to use:

- R1 = 46.4kΩ
- R2 = 84.5kΩ

Based on the same table, these values seem to correspond to a code of 0x006.

Could you clarify:

- Which resistor values (R1 and R2) should be used to encode the value 0x007?
- How should the reference voltages be determined for these codes, particularly for ensuring accuracy in setting the ver/rev?

Best regards,
Karthik

KarthiAru, 2024-12-09:
@mrpollo @davids5 Would you mind taking a moment to review this query and confirm the correct resistor values and reference voltages? Your guidance would be greatly appreciated. Thank you!

closed 2024-12-09 by KarthiAru

mrpollo, 2024-12-09:
Non Zero-One link https://github.com/pixhawk/Pixhawk-Standards/blob/master/DS-019%20Pixhawk%20versions%20and%20revisions.md